Raabi91/moonraker-telegram #17: Is it possible to use a single telegram bot with multiple chats?

prafols, 2021-03-16:
Moonraker-telegram runs fine with my phone and my Telegram account (chat id). But, I would like to have the possibility to allow my friends to control the same printer from their phones. So, I would like to be able to add more chat ids to the moonraker-telegram config. Is that possible?

Thanks and good work

excessnet, 2021-03-16:
Maybe create a group ? Didnt try, but maybe that group have a "unique" Chat ID you could use ?

prafols, 2021-03-17:
I've tried but didn't work. Apparently group chat ID's have a negative sign before. I don't know if this is a problem but it didn't work.

Raabi91, 2021-03-17:
at the moment is only one chat allowed. but i can do it in my list

CODeRUS, 2021-03-27:
> 
> 
> I've tried but didn't work. Apparently group chat ID's have a negative sign before. I don't know if this is a problem but it didn't work.

Group chat id works fine for me, You need to add bot to group first :)

prafols, 2021-04-30:
I tried it again without luck. Even with a fresh install it is not working for me. I added the bot and myself to the telegram group but it's not working. Can you provide any clues on how you managed to get it working? Thanks

Raabi91, 2021-05-01:
> I tried it again without luck. Even with a fresh install it is not working for me. I added the bot and myself to the telegram group but it's not working. Can you provide any clues on how you managed to get it working? Thanks

im working on it atm.

prafols, 2021-05-01:
Hey, I managed to get it working.

Basically I've modified two lines in the bot.py file and now it works. Take a look at my attachment (attached here within a zip)

Cheers

[bot.zip](https://github.com/Raabi91/moonraker-telegram/files/6409757/bot.zip)

Raabi91, 2021-05-01:
> Hey, I managed to get it working.
> 
> Basically I've modified two lines in the bot.py file and now it works. Take a look at my attachment (attached here within a zip)
> 
> Cheers
> 
> [bot.zip](https://github.com/Raabi91/moonraker-telegram/files/6409757/bot.zip)

Yes, that's enough, but you can't currently use comandos with the inline keyboard.

I already have the modified bot.py with all functions, but I am still testing it.

Raabi91, 2021-05-09:
@prafols with the last update should all working now

closed 2021-05-09 by Raabi91