Raabi91/moonraker-telegram #21: Minor issue with locale

melvinisken, 2021-03-23:
Hi! I just encountered a small issue with the telegram.sh script. I use a localized environment (german) which seems to conflict with printf formatting. Everytime the script is run, I get the error: 
moonraker-telegram/scripts/telegram.sh: Zeile 31: printf: 0.0: Ungültige Zahl. (line 31; invalid number)
I could cure this error by adding 
$ LC_NUMERIC="en_GB.UTF-8" 
at the start of line 31 (see https://stackoverflow.com/questions/12845997/printf-command-inside-a-script-returns-invalid-number).
I don't know if this behaviour might cause other bugs so I wanted to let you know. At least it seems that the script itself is still working, but so far I only tested it with the printer beeing offline.

Raabi91, 2021-03-24:
welche deutsche Environment nutzt du den ich nutz die nämlich auch und habe keine probleme

melvinisken, 2021-03-24:
locale gibt folgendes aus:
`LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=`
Ich habe die Einstellungen mit raspi-config gesetzt. Ansonsten ist es ein Pi 3 Model B Rev. 1.2, Raspbian Buster, alle Updates installiert.

Raabi91, 2021-03-25:
genau das selbe System fahr ich auch.

kannst du das locale im script mal entfernen und mir mal den Auszug davon schicken wenn du es mit  

sh /home/pi/moonraker-telegram/scripts/telegram.sh "2"

ausführst??

danke

melvinisken, 2021-03-26:
Jetzt hatte ich Zeit es zu testen. "sh" zu Beginn der Zeile macht den Unterschied.

```
$ sh /home/pi/moonraker-telegram/scripts/telegram.sh "2"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  162k    0  162k    0     0  2008k      0 --:--:-- --:--:-- --:--:-- 2008k
{"ok":true,"result":{"message_id":24,"from":{[...] file_size":39928,"width":640,"height":480}],"caption":"Finished printing CCR10_Laufrolle.gcode"}}

$ /home/pi/moonraker-telegram/scripts/telegram.sh "2"
/home/pi/moonraker-telegram/scripts/telegram.sh: Zeile 31: printf: 0.0: Ungültige Zahl.
/home/pi/moonraker-telegram/scripts/telegram.sh: Zeile 36: printf: .0561628186857900500: Ungültige Zahl.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  166k    0  166k    0     0  1132k      0 --:--:-- --:--:-- --:--:-- 1132k
{"ok":true,"result":{"message_id":25,"from":{ [...] file_size":43078,"width":640,"height":480}],"caption":"Finished printing CCR10_Laufrolle.gcode"}}
```
(den Ausgabestring habe ich gekürzt). Ich nehme an, das liegt also daran, dass dein Script die sh-shell nutzen muss um korrekt zu funktionieren, nicht die bash-shell. Der Fehler verschindet auch, wenn man das #!/bin/bash bei telegram.sh durch #!/bin/sh ersetzt. Dann kann ich es auch direkt (ohne sh) aufrufen und es kommt kein Fehler.

Raabi91, 2021-03-26:
okay dann ersetzt ich das #!/bin/sh am Anfang dann sollte ja alles in Zukunft passen bei dir

melvinisken, 2021-03-26:
Alles klar, danke. Aber funktioniert hat es die ganze Zeit trotzdem (es wurde eine Nachricht an per Telegram verschickt), nur kam halt die Meldung beim manuellen ausführen. Wenn das über Klipper läuft sieht man den Fehler ja auch sowieso nicht :-) Ich dachte nur, ich erwähne es, falls es doch mal irgendwann Probleme machen sollte.

Raabi91, 2021-03-26:
wenn es über klipper läuft wird es auch immer per sh ausgeführt und nicht ohne sh oder mit bash aber ja danke für den hinweiß. 

dann mach ich hier mal zu da fixed

closed 2021-03-26 by Raabi91