Raabi91/moonraker-telegram #27: Adding status message variables in telegram_config.sh

alexcroce, 2021-04-06:
When i send /state command to the bot, there are several variables. But not all of them are configurable via telegram_config.sh.

Here we can see 4 hard coded messages:

```
# #elif [ "$state_msg" = "5" ]; then
    if [ "$print_state_read1" = "printing" ]; then    
        msg="$msg_state"
    elif [ "$print_state_read1" = "standby" ]; then    
        msg="hey, i'm idling, please let me print something"
    elif [ "$print_state_read1" = "complete" ]; then    
        msg="hey, i finished the last print now i am idling"
    elif [ "$print_state_read1" = "paused" ]; then    
        msg="hey, i'm on break, please take a look"
    elif [ "$print_state_read1" = "error" ]; then    
        msg="hey, i had a error. please look it up"
     fi
```

I ask the possibility to change them via telegram_config.sh

Thx in advance.

Raabi91, 2021-04-10:
when i have time i will include it

Raabi91, 2021-04-29:
so the new variables are now there. after the update, however, they have to be added themselves.

https://github.com/Raabi91/moonraker-telegram/blob/master/docs/FAQ.md#new-variables-are-not-displayed-in-my-config

closed 2021-04-29 by Raabi91

alexcroce, 2021-04-29:
Great! Thx!