Raabi91/moonraker-telegram #28: cannot got replied at all

tjengbudi, 2021-04-10:
hello,
i installed the script and configure the moonraker_telegram config and add the bot token and the chat id
when the service running, my message got read by the script. but it never got replied. is there any log that i can check?
if based on the link api.telegram,org status i see that my text already read. but i think it is not processed
so i believe the config of thebot is correct

i try to send /state command but nothing happend or replied

tjengbudi, 2021-04-10:
tested telegram replied with 
sh /home/octo/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "0"

and telegram got the message. so i think it something got failure but not know where..

i check that /help and /print got replied from the scripts. but when printer is printing there is no notification send to the bot.

/state still did not respond. 
i check in the moonraker_telegram have websocket_state.txt this contain state of my printer. but never send to the telegram client

i check in the script folder, there is state_config.txt what this file used for?

Raabi91, 2021-04-10:
try sh /home/octo/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "1"

if there is no message then check your webcam settings in moonraker-telegram.

a log I have at the moment not yet but should come

tjengbudi, 2021-04-10:
`/home/octoprint/moonraker-telegram/scripts/telegram.sh: 103: /home/octoprint/moonraker-telegram/scripts/telegram.sh: convert: not found
`

i did not set the webcam yet

Raabi91, 2021-04-10:
The disable the Webcam in your telegram_config

tjengbudi, 2021-04-10:
emm noted ... already disabled and got replied after restart. will try while printing.

thanks for help

note: it will be nice if there is some explanation about this in doc. it will much help for newbie like me.
thanks btw

closed 2021-04-10 by tjengbudi