Raabi91/moonraker-telegram #35: Turn Light on

Sven9719, 2021-04-17:
Hello, is it possible before sending a status with image to turn on a relay that is included in Moonraker and turn it off again after recording?

Raabi91, 2021-04-25:
hey and sorry for the late write back had privately very much around the ears. at the moment it is not possible but I can see if you can do something there

Raabi91, 2021-04-26:
@Sven9719 
i have now two new comands in the last update
first on ist /gcode_macro if you have your leds over a gcode_macro
and the second one ist /power if you use the leds over the powerplugin

pleas let me know when it works

Sven9719, 2021-04-28:
Hey Raabi,
the extension works perfectly, but I imagined that the LEDs would be switched on automatically before the picture is taken and then switched off again automatically afterwards.

Best regards

Raabi91, 2021-04-29:
ah so there I have to look at how I could implement something like that run your leds via klipper or via moonraker?

Sven9719, 2021-04-29:
They run with the moonraker Power Plugin, thank you so much 🙏

excessnet, 2021-04-29:
@Sven9719 I didnt test it, but it should be something like this.

Make two macro for the power plugin on _printer.cfg_, something like this :

```
[gcode_macro LED_ON]
gcode:
  {action_call_remote_method("set_device_power",
                             device="led_psu",
                             state="on")}

[gcode_macro LED_OFF]
gcode:
  {action_call_remote_method("set_device_power",
                             device="led_psu",
                             state="off")}
```

then, on telegram, you just send the two commands

```
/gcode_macro LED_ON
/gcode_macro LED_OFF
```

Sven9719, 2021-04-29:
But then I have to keep doing it myself and it doesn't happen automatically, right?

Raabi91, 2021-04-29:
> @Sven9719 I didnt test it, but it should be something like this.
>
> Make two macro for the power plugin on _printer.cfg_, something like this :
>
> ```
> [gcode_macro LED_ON]
> gcode: [...]
 
you only need to use /gcode_macro then you will asked with an inlinekeyboard

@Sven9719 okay i will take a look at this to do this automatically for the picture

Raabi91, 2021-05-13:
> But then I have to keep doing it myself and it doesn't happen automatically, right?

I hope it works without any problems because unfortunately I cannot test it intensively.

Please try it out

https://github.com/Raabi91/moonraker-telegram/blob/master/docs/FAQ.md#how-to-use-automatic-led-for-cam

closed 2021-06-03 by Raabi91

digitalhandwerker, 2022-01-18:
I have a follow-up request/bug(?):

I set up LED and it is working as long as webcamd is running.
My problem is that I bound the LED and webcamd together (moonraker power plugin's `bound_service:`) so that the webcam is only running when I want to (no light, no camera needed).
But this is not working with the bot. I always get "webcam link not working image". (I tried setting the delay to 10s but still not working).

Moonraker should handle the starting and stopping of webcamd. But I think the bot is trying fetch the snapshot url before the service is running. The delay seems to start after a first fetch.
I guess this is unintended behavior?

Thanks.