Raabi91/moonraker-telegram #49: Not receiving any messages, but I can send message manually

simpat1zq, 2021-06-01:
Thanks for this plugin. I just installed it, and I can send a message using this command:

sh /home/pi/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "0"

And I've confirmed the service is running. But, when I send a message back with '/state' and any of the other commands, it doesn't reply back. How do I troubleshoot this?

simpat1zq, 2021-06-01:
Nevermind. I'm guessing /state doesn't work when idle. /help worked. I do have another question, is it possible to have it only send messages while printing? I would like a message every 30 minutes, but only while printing.

Raabi91, 2021-06-01:
do you have setup the telergam plugin?? like time and cam ....

simpat1zq, 2021-06-01:
Actually, it looks like /help works consistently, but a lot of the other commands are very hit or miss. /state never works. /power only returns 'Your List 1'. I have time:1800, but I only received a message while it was idle, and haven't received one since then. Everything else for the plugin is setup.

Raabi91, 2021-06-01:
try this 

sh /home/pi/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "1"

when this not worked then check your cam setup for telegram

simpat1zq, 2021-06-01:
This was working when first setup as I mentioned in my first post, but now it's giving me an error: 

`pi@elf:~ $ sh /home/pi/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "1"
Runtime error (func=(main), adr=28): Divide by zero

-2408.006272153929
pi@elf:~ $ Runtime error (func=(main), adr=28): Divide by zero

-2408.257093664957
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Failed to create the file
Warning: /home/pi/moonraker-telegram/picture/cam_new.jpg: Permission denied
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (23) Failed writing body (0 != 1442)
convert-im6.q16: Not a JPEG file: starts with 0x2d 0x2d `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/jpeg.c/JPEGErrorHandler/335.
convert-im6.q16: no images defined `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/convert.c/ConvertImageCommand/3258.
{"ok":false,"error_code":413,"description":"Request Entity Too Large"}/home/pi/moonraker-telegram/scripts/telegram_custom.sh: 98: /home/pi/moonraker-telegram/scripts/telegram_custom.sh: Syntax error: "fi" unexpected
^C
pi@elf:~ $
`

Raabi91, 2021-06-01:
what you will get when you call yourip/printer/objects/query?display_status in browser

simpat1zq, 2021-06-01:
result | &nbsp;
-- | --
status | &nbsp;
display_status | &nbsp;
progress | 0
message | null
eventtime | 865842.154464082

Raabi91, 2021-06-01:
looking good

and here http://yourip/printer/objects/query?print_stats

simpat1zq, 2021-06-01:
status | &nbsp;
-- | --
print_stats | &nbsp;
print_duration | 3367.426450329949
total_duration | 3419.8622189559974
filament_used | 10240.010040000003
filename | "TTRTrayx4.gcode"
state | "printing"
message | ""
eventtime | 866154.04229089

Raabi91, 2021-06-01:
looks good to let me see your config without your chat id an token

simpat1zq, 2021-06-01:
#!/bin/bash
# Edit only the variables between the ""
# Port from moonraker
port="7125"
# Your telegram bot token
token="xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"
# Your chat ID
chatid="xxxxxxxx"
#
# messages for the states
### for the placeholders in the messages look at https://github.com/Raabi91/moonraker-telegram/blob/master/docs/Variables.md
#
# Start message
msg_start="Started printing $print_filename"
# error/failed message
msg_error="printing of $print_filename Failed"
# Pause message
msg_pause="printing of $print_filename Paused"
# Complete message
msg_end="Finished printing $print_filename"
# Time message
msg_state="Printing $print_filename at $print_progress. Current Time $print_current. Remaining Time $print_remaining"
# standby message
#msg_standby="hey, i'm idling, please let me print something"
# complete message
msg_complete="hey, i finished the last print now i am idling"
# paused message
msg_paused="hey, i'm on break, please take a look"
# error message
msg_error="hey, i had a error. please look it up"
#
# time in seconds to get an State update. to disable set it to 0
time="1800"
# with picture = 1, without picture = 0  
picture="1"
# your webcam snapshot link
webcam="http://192.168.45.104:8081/"
#rotate the pic bevor sending, use degrease 0-360
rotate="0"
#flip the pic horizontally bevor sending, 1 = yes, 0 = No
horizontally="0"
#flip the pic vertically bevor sending, 1 = yes, 0 = No
vertically="0"
# Make all commands Disable with 1
bot_disable="0"
# delay for the Print start Message
delay_start_msg=="0"
# delay for the Print end Message
delay_end_msg=="0"
# Delay for the Pause Message
delay_pause_msg=="0"
#Led on link for picture
led_on=""
#Led on wait time before picture is taken (in seconds) 
led_on_delay="0"
#Led off link for picture
led_off=""
#Led off wait time after picture is taken (in seconds)
led_off_delay="0"

Raabi91, 2021-06-01:
your webcam link is false

Raabi91, 2021-06-01:
it musst looking like this http://your ip:8081/?action=snapshot

simpat1zq, 2021-06-01:
Thanks. I'm using MotionEye to connect to my RTSP cam. I accidentally put the streaming URL instead of the still image URL. (http://192.168.45.104:8123/api/hassio_ingress/xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx/picture/1/current/)

But I fixed that now, and I still get the same error. Going to that URL gives me a jpg file.

Raabi91, 2021-06-01:
i have no rtsp cam here to test

simpat1zq, 2021-06-01:
You don't need one. I'm just connecting straight to MontionEye at the URL in my last message. That URL just give me a .jpg image. 

And then I get this error. Looks like it's actually a little different than the first:

```
pi@elf:~ $ sh /home/pi/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "1"
Runtime error (func=(main), adr=28): Divide by zero

-4272.188119099941
pi@elf:~ $ Runtime error (func=(main), adr=28): Divide by zero

-4272.438900296926
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Failed to create the file
Warning: /home/pi/moonraker-telegram/picture/cam_new.jpg: Permission denied
100    17  100    17    0     0   1545      0 --:--:-- --:--:-- --:--:--  1545
curl: (23) Failed writing body (0 != 17)
convert-im6.q16: Not a JPEG file: starts with 0x2d 0x2d `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/jpeg.c/JPEGErrorHandler/335.
convert-im6.q16: no images defined `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/convert.c/ConvertImageCommand/3258.
{"ok":false,"error_code":413,"description":"Request Entity Too Large"}/home/pi/moonraker-telegram/scripts/telegram_custom.sh: 98: /home/pi/moonraker-telegram/scripts/telegram_custom.sh: Syntax error: "fi" unexpected

```

Raabi91, 2021-06-01:
curl -o cam_new.jpg your_webcam_link 

what will do this in your home folder?

Raabi91, 2021-06-01:
do you have try to reboot your pi?? it can be possible that he still records the old stream in the Hintergung. since the bottle address was entered

simpat1zq, 2021-06-01:
Ah actually, it looks like that was it. I was able to access the still image from my computer, but not from the pi. It was looking for an auth. Looks like I'll have to disable the cam for this plugin unless I can figure that out.

simpat1zq, 2021-06-02:
I'm trying to figure out some workarounds for my camera, but do you think it might be a good idea to try and have it send out a message even if it can't retrieve the image? There might be times when it can't grab the image.

simpat1zq, 2021-06-02:
Also, I just got everything figured out with Motioneye, and now the URL I have configured returns a jpg image, but the telegram plugin doesn't appear to work. There's a new error message:  

```
pi@elf:~ $ sh /home/pi/moonraker-telegram/scripts/telegram.sh "this is a message from your printer" "1"
7376
3245.319197096047
pi@elf:~ $ 7376
3245.068212044076
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0Warning: Failed to create the file
Warning: /home/pi/moonraker-telegram/picture/cam_new.jpg: Permission denied
  0  763k    0  2573    0     0  88724      0  0:00:08 --:--:--  0:00:08 88724
curl: (23) Failed writing body (0 != 2573)
convert-im6.q16: insufficient image data in file `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/jpeg.c/ReadJPEGImage/1117.
convert-im6.q16: no images defined `/home/pi/moonraker-telegram/picture/cam_new.jpg' @ error/convert.c/ConvertImageCommand/3258.
{"ok":false,"error_code":400,"description":"Bad Request: IMAGE_PROCESS_FAILED"}/home/pi/moonraker-telegram/scripts/telegram_custom.sh: 98: /home/pi/moonraker-telegram/scripts/telegram_custom.sh: Syntax error: "fi" unexpected
^C
pi@elf:~ $ wget http://192.168.45.104:9000/picture/1/current/
--2021-06-02 13:22:32--  http://192.168.45.104:9000/picture/1/current/
Connecting to 192.168.45.104:9000... connected.
HTTP request sent, awaiting response... 200 OK
Length: 758739 (741K) [image/jpeg]
Saving to: ‘index.html’

index.html          100%[===================>] 740.96K  --.-KB/s    in 0.1s

2021-06-02 13:22:32 (6.95 MB/s) - ‘index.html’ saved [758739/758739]

pi@elf:~ $

```

simpat1zq, 2021-06-02:
IT WORKED!!! I just had to delete the cam_new.jpg file. I'm guessing that the corrupted file from the previous attempts was causing problems. I did get a divide by zero error when running the command, but it still worked:
```
pi@elf:~/moonraker-telegram/picture $ Runtime error (func=(main), adr=28): Divide by zero

-4601.461539702956
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  753k  100  753k    0     0   671k      0  0:00:01  0:00:01 --:--:--  671k
{"ok":true,"result":{.......

```

Raabi91, 2021-06-03:
if the message now comes with a picture, the error Divide by zero does not matter for the time being. it will take care of itself when moonraker provides other data.

closed 2021-06-10 by Raabi91