Raabi91/moonraker-telegram #57: On Fluidd no answer and no notification, only: "Your list 1"

dwolf91, 2021-07-05:
Dear Friend!

I Have a issue with Fluidd. I Have 2 printer 1. is with Mainsail and 1 is with Fluidd. In Mainsail the Script work fine, with no problem, But in Fluidd the printer send me nothing. Only work the /help and /gif command, the /state make nothing, and the another commands becomme a "Your list 1" answer.

Thanks the help and sorry by bad Englisch.

Raabi91, 2021-07-05:
ist not a problem with fluidd. fluidd and mainsail ar only webinterfaces. the plugin only connect to moonraker. when messages come and comands are wokring than your bot is working. I think is a problem wit your moonraker connection do you have the setup the right port for moonraker?

dwolf91, 2021-07-05:
> ist not a problem with fluidd. fluidd and mainsail ar only webinterfaces. the plugin only connect to moonraker. when messages come and comands are wokring than your bot is working. I think is a problem wit your moonraker connection do you have the setup the right port for moonraker?

I think so. What i see in Install is an error on "fluiddpi.local" but i don't now why. I think i try to uninstall the script and then install again.

dwolf91, 2021-07-06:
Ok now It's Work fine. I Have complet reistalld my rpi4 with fluidd and now work! Sorry for disturbance! You can Close/delete the issue, Thanks, and sorry!

closed 2021-07-09 by Raabi91