Raabi91/moonraker-telegram #58: Mainsail no notification, only: "Your list 1"

cosmo87-1, 2021-08-01:
I don't get any response in Telegram except for "/ Help" and "/ gif".
In my old printer.cfg I still have a "Shell command". At least "/ state" works with that.
Does the printed.cfg have to be adapted for the current version?

German:
Ich bekomme keine Reaktion in Telegram außer bei "/Help" und "/gif".
In meiner alten printer.cfg habe ich noch ein "Shell command". Damit funktioniert immerhin "/state".
Muss für die aktuelle Version die printed.cfg angepasst werden?

Raabi91, 2021-08-07:
sorry für das späte melden war im moment sehr im stress. selbst bei der letzten version sollte immer noch alles so laufen wie vorher ohne das du was ändern musst.

die frage ist jetzt nur funktioniert übers Shell command auch die bilder?

cosmo87-1, 2021-08-09:
Das mit dem Stress kenne ich zu gut ;)
Ich fühle mich mit dem "Shell command" gerade richtig doof. Meinst du: sh /home/pi/moonraker-telegram/scripts/telegram.sh?

Meine printer.cfg:
[gcode_shell_command telegram]
command: sh /home/pi/moonraker-telegram/scripts/telegram.sh "print done! /state"
timeout: 2.
verbose: false

Meine Makros:
alt:
[gcode_macro telegram]
gcode:
    RUN_SHELL_COMMAND CMD=telegram

Neu:
[gcode_macro telegram_MSG]
gcode:
    RESPOND PREFIX=telegram: MSG="Nachricht"

[gcode_macro telegram_MSG+picture]
gcode:
    RESPOND PREFIX=telegram_picture: MSG="Nachricht mit Bild"

Wenn ich in Telegramm /state schreibe kommt nichts.

Raabi91, 2021-08-14:
dann hört sich das so an als würde sich der bot aufhängen

mal in ssh einlogen

dann mal

sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_stop.sh

ausführen und dann das ausführen

sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_start.sh

anschließend sendest du /state und sendest mir mal die Ausgabe die dann erscheint. aber Achtung deine ids sind darin ersichtlich

cosmo87-1, 2021-08-14:
folgendes wird mir dann in SSH angezeigt:
(IDs und namen geändert)
```
pi@raspberrypi:~ $ sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_stop.sh
kill: (925): Kein passender Prozess gefunden
Getötet
pi@raspberrypi:~ $ sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_start.sh
pi@raspberrypi:~ $ Handshake status 401 Unauthorized
### closed ###
Retry : Sat Aug 14 09:49:04 2021
I am listening ...
text
123456789
Got chat: 123456789
chat  id: 123456789
Got command: /state
chat id are same
(standard_in) 1: syntax error
(standard_in) 2: syntax error

(standard_in) 1: syntax error
Handshake status 401 Unauthorized
### closed ###
Retry : Sat Aug 14 09:49:14 2021
text
123456789
Got chat: 123456789
chat  id: 123456789
Got command: /help
chat id are same
(standard_in) 1: syntax error
(standard_in) 2: syntax error

(standard_in) 1: syntax error
{"ok":true,"result":{"message_id":123,"from":{"id":123456789,"is_bot":true,"first_name":"sapphire_bot","username":"sapphire_XX_bot"},"chat":{"id":123456789,"first_name":"xxx","type":"private"},"date":1628927359,"text":"available commands are:\n  /state - Sends the current status including a current photo.\n  /pause - Pause current Print.  A confirmation is required\n  /resume - resume current Print.\n  /cancel - Aborts the currently running print. A confirmation is required\n  /help - show list of commands.\n  /print - Will open a file dialog showing the files stored in moonraker. You can select a file to print it.\n  /power - Will open a file dialog showing the Power devices of moonraker. You can choose a device to interact with it\n  /gcode_macro - Will open a file dialog showing the custom Gcode_macros of the printer.cfg. You can choose a macro to execute it\n  /gif - send a 5 second gif","entities":[{"offset":26,"length":6,"type":"bot_command"},{"offset":89,"length":6,"type":"bot_command"},{"offset":149,"length":7,"type":"bot_command"},{"offset":183,"length":7,"type":"bot_command"},{"offset":258,"length":5,"type":"bot_command"},{"offset":291,"length":6,"type":"bot_command"},{"offset":400,"length":6,"type":"bot_command"},{"offset":519,"length":12,"type":"bot_command"},{"offset":649,"length":4,"type":"bot_command"}]}}Handshake status 401 Unauthorized
### closed ###
Retry : Sat Aug 14 09:49:24 2021
Handshake status 401 Unauthorized
### closed ###
Retry : Sat Aug 14 09:49:34 2021

```

da scheint wirklich etwas mit dem Bot nicht zu stimmen.
Soll ich einen neuen erstellen und alles auf dem Pi deinstallieren was mit telegram zu tun hat?

Raabi91, 2021-08-14:
Unauthorized deutet auf moonraker hin wie schaut deine config dort aus

cosmo87-1, 2021-08-14:
die moonraker.conf?

```
[server]
host: 0.0.0.0 
port: 7125
klippy_uds_address: /tmp/klippy_uds
enable_debug_logging: True 
config_path: ~/klipper_config 

[authorization]
trusted_clients:
    10.0.2.20 
    10.0.2.0/24
    10.0.0.0/24

[octoprint_compat]

[update_manager]
enable_repo_debug: False
enable_auto_refresh: False

[update_manager client mainsail]
type: web
repo: meteyou/mainsail
path: ~/mainsail

[update_manager client moonraker-telegram]
type: git_repo
path: /home/pi/moonraker-telegram
origin: https://github.com/Raabi91/moonraker-telegram.git
env: /home/pi/.moonraker-telegram-env/bin/python
requirements: scripts/moonraker-telegram-requirements.txt
install_script: scripts/install.sh

```

Raabi91, 2021-08-14:
füge mal bei trusted_clients: noch
127.0.0.0/8
ein

cosmo87-1, 2021-08-14:
Jetzt funktioniert es! Danke!
Fehlt nur noch das Bild.

Raabi91, 2021-08-14:
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#respond

das noch in deine printer.cfg

cosmo87-1, 2021-08-14:
ja [respond] reicht aus :)
jetzt funktioniert auch das! 
Habe ich das alles in der Installation überlesen?!

"/state" bringt noch kein Bild

Raabi91, 2021-08-14:
super dann mach ich mal zu

closed 2021-08-14 by Raabi91