Raabi91/moonraker-telegram #61: Moonraker log: Unauthorized, [No User]

DrCoolHands, 2021-09-08:
Hello,

I setup the moonraker-telegram bot on FluidOS Pi image and get the following errors in the Moonraker.log:

`2021-09-07 21:21:47,099 [web.py:log_exception()] - 401 GET /websocket (127.0.0.1): Unauthorized
2021-09-07 21:21:47,102 [app.py:log_request()] - 401 GET /websocket (127.0.0.1) [No User] 5.73ms
2021-09-07 21:21:57,123 [web.py:log_exception()] - 401 GET /websocket (127.0.0.1): Unauthorized
2021-09-07 21:21:57,126 [app.py:log_request()] - 401 GET /websocket (127.0.0.1) [No User] 6.50ms
2021-09-07 21:22:07,147 [web.py:log_exception()] - 401 GET /websocket (127.0.0.1): Unauthorized
2021-09-07 21:22:07,150 [app.py:log_request()] - 401 GET /websocket (127.0.0.1) [No User] 5.49ms`

I am using user authentication in Fluidd, not sure how that works back with Moonraker.

Raabi91, 2021-09-08:
Moonraker telegram does not currently work with user authentication. But since I'm currently in exam stress, I unfortunately don't have time to integrate it at the moment.

Raabi91, 2021-09-18:
with the last update the whole thing should now be possible via the api key as far as i could test it

DrCoolHands, 2021-09-18:
I can confirm that the added API key works with force login = true.
Thank you!

closed 2021-09-18 by DrCoolHands