Raabi91/moonraker-telegram #63: Request: Several improvements

petitxef, 2021-09-16:
Hi!

Just thinking some cool improvement that will be nice to have:

1. Shutdown/reboot the system
2. Visualize the result of a Macro/query on telegram (now just on the console)
3. Tune the bed/extruder temp on telegram (not macros), as Octoprint-Telegram plugin

Thanks!

Raabi91, 2021-09-18:
for 1. I have now inserted the /host command

at 2. be a bit more precise if you only want to know that your filament sensor has triggered then you only have to execute a custom msg with the gcode

the 3. I will still include

and 4. is already present in the variables

petitxef, 2021-09-18:
First of all, Thank you for your efforts!

the 2nd. just thinking about see on telegram the result of the macro (as the console, but remotely on telegram)

On the 4th, I'll need to check how to do it :)

Thanks a lot!

Raabi91, 2022-02-17:
so 3. is intigratet

and for second i thinks is the best when you will call custom messages at the gcode

closed 2022-03-05 by Raabi91