Raabi91/moonraker-telegram #68: [Question] led_on / led_off multiple

Jack77777777, 2021-11-16:
1. Can i setup more than one led to power on/off ?
2. Is there a command for a snapshot/picture, not gif ? If not i can configure it with own custom MSG.

Raabi91, 2021-11-16:
1. if u use the monraker power plugin or gcode macro sure

2. an snapshot comand is atm not avaible it comes in the next update

Jack77777777, 2021-11-17:
If i set up this way:
led_on="http://127.0.0.1:$port/machine/device_power/on?FRAME_LIGHT"
led_on="http://127.0.0.1:$port/printer/gcode/script?script=POWER_ON_CAMLIGHT"
only the second executes. When commenting out the second, the first executes.
How can i have both executed ?

Raabi91, 2021-11-17:
with a custom gcode macro in klipper like this

[gcode_macro LEDS_ON_FOR_CAM]
gcode:
 POWER_ON_CAMLIGHT
 {action_call_remote_method("set_device_power",
                             device="FRAME_LIGHT"",
                             state="on")}
                             
then put this in the telegram config:
http://127.0.0.1:$port/printer/gcode/script?script=LEDS_ON_FOR_CAM"

Jack77777777, 2021-11-17:
I am so dump. Closing this.

closed 2021-11-17 by Jack77777777