Raabi91/moonraker-telegram #69: Telegram not getting answer

sharkyenergy, 2021-12-11:
Hello!

first off, thank you for your precious work. I installed it today and got it working, but after 1 hour it didnt work anymore. I did not change anything in the meantime. I send /state and get no answer.

this is the output:
```

text
130735587
Got chat: 130735587
chat  id: 130735587
Got command: /state
chat id are same
(standard_in) 1: syntax error
(standard_in) 2: syntax error

(standard_in) 1: syntax error
Handshake status 401 Unauthorized
### closed ###
Retry : Sat Dec 11 19:54:08 2021
Handshake status 401 Unauthorized
### closed ###
```

these are my trusted clients in moonracker config:

```

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  FE80::/10
  ::1/128
```

and I already added [respond] to my printer config.

Any Idea of what might be wrong? Already tried restarting, no luck.

thank you have a nice day

Raabi91, 2021-12-12:
did you use fluidd or mainsail? if u use fluidd is force_logins True in moonraker??

sharkyenergy, 2021-12-12:
I am using fluidd and yes, force logins is set to true because I need to login from remote.

sharkyenergy, 2021-12-12:
I found the part about the api key in the telegram config. added the api key and now it works properly! thank you very much!

closed 2021-12-12 by Raabi91