Raabi91/moonraker-telegram #70: Sent command but no response

eskiesirius, 2021-12-12:
I can receive message from the printer but when i send a command,there is no reply from the printer.

Raabi91, 2021-12-12:
go into ssH
TYPE
sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_stop.sh

THEN TYPE
sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_start.sh

then let me see the output when you have send a comand

eskiesirius, 2021-12-12:
i got this error 

```
KeyError: "No suggested keys ['message', 'edited_message', 'channel_post', 'edited_channel_post', 'callback_query', 'inline_query', 'chosen_inline_result', 'shipping_query', 'pre_checkout_query'] in {'update_id': <update_id>, 'my_chat_member': {'chat': {'id': <group_id>, 'title': '<title>', 'type': 'group', 'all_members_are_administrators': True}, 'from': {'id': <>, 'is_bot': False, 'first_name': 'ashdread', 'username': '<>'}, 'date': 1639293599, 'old_chat_member': {'user': {'id': <bot id>, 'is_bot': True, 'first_name': 'AlfredBot', 'username': '<bot_username>'}, 'status': 'left'}, 'new_chat_member': {'user': {'id': <bot id>, 'is_bot': True, 'first_name': 'AlfredBot', 'username': '<bot_username>'}, 'status': 'member'}}}"
Traceback (most recent call last):
  File "/root/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/loop.py", line 67, in run_forever
    self._update_handler(update)
  File "/root/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/loop.py", line 153, in <lambda>
    collectloop.input_queue.put(_extract_message(update)[1]))
  File "/root/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/loop.py", line 111, in _extract_message
    'pre_checkout_query'])
  File "/root/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/__init__.py", line 68, in _find_first_key
    raise KeyError('No suggested keys %s in %s' % (str(keys), str(d)))
```

Raabi91, 2021-12-12:
okay try the following.

delete your bot and create it again, but when creating the bot do not write in the private chat but invite directly in the group chat and start the bot there with /start first

eskiesirius, 2021-12-12:
should i mention the bot?

eskiesirius, 2021-12-12:
i got unauthorized

eskiesirius, 2021-12-12:
same error

eskiesirius, 2021-12-12:
i set the chat_id to the group id..

Raabi91, 2021-12-12:
i have pushed an update to github please update it once on your pi and try if it works then

holterb, 2021-12-29:
i also get no response, and here is what i get when i run the start.sh (i have just gotten all the latest updates)

File "/home/pi/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/loop.py", line 59, in run_forever
    allowed_updates=allowed_updates)
  File "/home/pi/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/__init__.py", line 1001, in getUpdates
    return self._api_request('getUpdates', _rectify(p))
  File "/home/pi/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/__init__.py", line 492, in _api_request
    return api.request((self._token, method, params, files), **kwargs)
  File "/home/pi/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/api.py", line 155, in request
    return _parse(r)
  File "/home/pi/.moonraker-telegram-env/lib/python3.7/site-packages/telepot/api.py", line 150, in _parse
    raise exception.TelegramError(description, error_code, data)
telepot.exception.TelegramError: ('Not Found', 404, {'ok': False, 'error_code': 404, 'description': 'Not Found'})
Traceback (most recent call last):

Raabi91, 2021-12-30:
> 

have you try to run the install.sh again with sudo after update?

closed 2022-02-17 by Raabi91