Raabi91/moonraker-telegram #71: Feature request

sharkyenergy, 2021-12-12:
Hello! 

amazing work you did! thank you very much!

I'd have some feature ideas I'd like to show you..

1) min time interval between automated messages.
 
I have set it up to send a message every 5%. On big prints this is a message every 10/20 minutes. on small prints it is a message every 30 seconds. It would be cool to define a timeout (for example 5 minutes) and all the AUTOMATIC messages (by % and by height) are ignored if the last message has been sent less than X minutes ago.

2) send message with image after first layer.

would be cool to have this as option, to see if there are any print adhesion problems.

3) set temperature via telegram

it would be cool to be able to adjust the temperature of the bed and nozzle via telegram.

4) be able to set fan and outputs via telegram.

that would allow to change the light brightness and fan speeds via telegram. 

5) be able to restart the host.

in case of any problem it would be cool to be able to restart the host from remote.

Raabi91, 2022-02-17:
3. and 5. are now live.

sharkyenergy, 2022-02-17:
Great thank you! Are there other points of my list that you plan to add? Will try these new features asap

mtmello2003, 2024-03-05:
Items (1) and (4) would be very useful...