Raabi91/moonraker-telegram #75: After Complete Installation, Fluidd Gives Error For Moonraker 'update_manager error'

Declan0811, 2021-12-22:
Basically the title, I have gone through the entire guide, but I am being hit by a ton of errors, and cannot figure out what to do. I have attached the log file, and any help would be greatly appreciated!
[moonraker (1).log](https://github.com/Raabi91/moonraker-telegram/files/7760088/moonraker.1.log)

Raabi91, 2021-12-22:
The log say that the virtuell env is missing. Ich got you any error at Installation?

Declan0811, 2021-12-22:
> The log say that the virtuell env is missing. Got you any error at Installation? 

No, I did not get any errors. Is there a way to reinstall everything, and try again?

Declan0811, 2021-12-23:
Nevermind, I just had to sudo when using './scripts/install.sh'

closed 2021-12-23 by Declan0811