Raabi91/moonraker-telegram #78: Question: Does `/state ` command reset countdown

TheLion, 2022-01-15:
Hi,

Does a manual `/state` command reset the countdown for the timed status messages?

TheLion, 2022-01-15:
I guess it doesn't. Thought this was the case, because I didn't receive any notifications efter I issues a `/state`. But turns out is doesn't sent any time-notifications. A manual message is delivered just fine though. The log doesn't show anything, except for

```
Sat 15 Jan 15:17:34 GMT 2022
Start moonraker-telegram
```
Will try a reboot after print is finished, maybe that clears things up.

Raabi91, 2022-01-16:
any new findings because here it is not so

TheLion, 2022-01-16:
Yes. It seemed like something had gone wrong. Just started a new print after a reboot and the notifications go just fine now. 

Any change you can add Pushover notifications, or create a new plugin?

Maybe through Shoutrrr ([https://containrrr.dev/shoutrrr/](https://containrrr.dev/shoutrrr/)), this also supports Telegram.

Raabi91, 2022-01-19:
> Yes. It seemed like something had gone wrong. Just started a new print after a reboot and the notifications go just fine now.
> 
> Any change you can add Pushover notifications, or create a new plugin?
> 
> Maybe through Shoutrrr (https://containrrr.dev/shoutrrr/), this also supports Telegram.

at the moment i have no time for it. that would be a completely new plugin. find currently quite little time on moonraker telegram to continue. as that is just so with work etc.

closed 2022-01-19 by Raabi91

TheLion, 2022-01-23:
> at the moment i have no time for it. that would be a completely new plugin. find currently quite little time on moonraker telegram to continue. as that is just so with work etc.

I understand.  No problem.