Raabi91/moonraker-telegram #79: Reporting wrong print time

digitalhandwerker, 2022-01-18:
Hey, thanks for the tool.

I installed it and I can write commands. The bot is always responding. Control command are working.
Interestingly in the `/state` command `$print_current` and `$print_remaining` do not match the times displayed in Mainsail.
(`$print_current` is always lower and `$print_remaining` is always higher.)
`$print_progress` is correct.

What can I do to help debugging?

Raabi91, 2022-01-19:
print current is simply drawn from moonraker on the print time from the first extrusion. therefore i have no influence on it. print remaining is calculated from print current and the print progress. mainsail uses the total time and partly lets other times flow in. due to the different processes at the printer, it can be different. for me, the start process only takes 4 minutes until the first extrusion, so it doesn't matter so much. 

I look but that I times as a second option offer

Translated with www.DeepL.com/Translator (free version)

zenturacp, 2022-03-05:
@Raabi91 well - print progress % is also way off from Mainsail? Is mainsail just doing some modifications to the calculations?

Raabi91, 2022-03-06:
> @Raabi91 well - print progress % is also way off from Mainsail? Is mainsail just doing some modifications to the calculations?

Yes, mainsail has its own calculation for the process and does not use the process from moonraker like the plugin . But I already have this on my to-do list.

Raabi91, 2022-04-15:
@zenturacp @digitalhandwerker 

new update is available with the fix and additional variables for the time display