Raabi91/moonraker-telegram #89: Plugin sends new and old screenshot

TheLion, 2022-03-05:
I just updated the plugin to v1.0.2-188-ge95e4d3 and it sends the correct screenshot, but also an old one, in my case the last one from a previous (not last) job. See screenshots from Telegram below.

[image]
[image]

Is there a cache/file I need to empty/delete?

Raabi91, 2022-03-05:
have published another update since I'm back home. should now no longer happen

TheLion, 2022-03-06:
Yes, I can confirm. Issue is solved with update.

closed 2022-03-06 by TheLion