Raabi91/moonraker-telegram #97: Unable to send messages with a picture.

DrCoolHands, 2022-03-28:
I have had moonraker-telegram working correctly for some time.
Since March 10th, it no longer will send messages with an attached picture. Text-only messages are sent correctly.

The log file has this associated error:
`Webcam[213.312 link has an error`

No changes to the config file have been made between the last working state and now.
The snapshot link is:
`webcam="http://127.0.0.1:8080/?action=snapshot"`
I can curl this address from the Pi, it is valid and contains an image.

Raabi91, 2022-03-31:
sry for the late replay

what will come when you use

curl --output /dev/null --fail -r 0-0  "http://127.0.0.1:8080/?action=snapshot"

DrCoolHands, 2022-03-31:
```
pi@unit-01:~ $ curl --output /dev/null --fail -r 0-0 "http://127.0.0.1:8080/?action=snapshot"
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 50192    0 50192    0     0  1531k      0 --:--:-- --:--:-- --:--:-- 1581k
```
No errors reported.

The picture-attached messages worked previously. I do not recall if I applied an update from the repo that instigated the issue.

Raabi91, 2022-04-02:
can you please install the latest update. if it is still then we need to go a little deeper

DrCoolHands, 2022-04-04:
Thanks. The latest update appears to have corrected the issue.

closed 2022-04-04 by DrCoolHands