Raabi91/moonraker-telegram #99: /gcode_macro returns only one of my visible macros

maze2013, 2022-04-02:
Since today the command "/gcode_macro" in Telegram only returns "Your list 1" and the macro "GET_TIMELAPSE_SETUP" (the most useless macro I have)
Everything worked fine for weeks now, but suddenly the problem showed up. 
I don't really get it. I've tried to restart moonraker-telegram with
sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_stop.sh and
sh /home/pi/moonraker-telegram/scripts/moonraker-telegram_start.sh
After about half an hour without any reaction on the ssh console I canceled the start script execution. 

If I restart the whole Raspberry Pi, the moonraker-telegram sends a nice picture with /state, a gif with /gif and everythind works except the /gcode_macro which still returns only one.

My moonraker.cfg in Mainsail OS:
````
[server]
host: 0.0.0.0
port: 7125
# Verbose logging used for debugging . Default False.
enable_debug_logging: False
# The maximum size allowed for a file upload (in MiB).  Default 1024 MiB
max_upload_size: 1024

[file_manager]
config_path: ~/klipper_config
log_path: ~/klipper_logs
# post processing for object cancel. Not recommended for low resource SBCs such as a Pi Zero. Default False 
enable_object_processing: False

[authorization]
cors_domains:
    https://my.mainsail.xyz
    http://my.mainsail.xyz
    http://*.local
    http://*.lan
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
refresh_interval: 168 

[update_manager client mainsail]
type: web
repo: mainsail-crew/mainsail
path: ~/mainsail

[update_manager client moonraker-telegram]
type: git_repo
path: /home/pi/moonraker-telegram
origin: https://github.com/Raabi91/moonraker-telegram.git
env: /home/pi/.moonraker-telegram-env/bin/python
requirements: scripts/moonraker-telegram-requirements.txt
install_script: scripts/install.sh

[update_manager timelapse]
type: git_repo
primary_branch: main
path: ~/moonraker-timelapse
origin: https://github.com/mainsail-crew/moonraker-timelapse.git

[timelapse]
````
Can you help me out there?

Raabi91, 2022-04-03:
i have uploaded an update. so it should work again now

maze2013, 2022-04-03:
Yes, it works again, thank you so much!

closed 2022-04-03 by maze2013