Raabi91/moonraker-telegram #106: send status updates in new messages

alexandrkut, 2022-05-27:
Hello!
Is it possible to send status updates in new messages (I did not find such an option in the config)?
I would like to have an archive of images of the printing process, not just the last image.

Raabi91, 2022-05-27:
What exactly do you mean by it each status message comes as a new message at the plugin

[image]

Raabi91, 2022-05-27:
[image]

alexandrkut, 2022-05-27:
> [image]

this is what I need!

My bot updates one message and at the end I see only the last image.
[image]

Raabi91, 2022-05-27:
I dont think you use this bot because you hav3 a timelpase message in you screen.

I think you use the moonraker-telegram-bot

alexandrkut, 2022-05-27:
Damn,
I accidentally installed both bots.

Sorry!

closed 2022-05-27 by alexandrkut