Raabi91/moonraker-telegram #109: Percentages stuck at 0 %

Stooovie, 2022-08-19:
Notifications are working correctly with one exception: the percentage is always stuck at zero. How can I help to fix this? 

klipper
v0.10.0-548-g2357221b

crowsnest
v3.0.1-2-g946fad1

moonraker
v0.7.1-626-g853de9e

moonraker-telegram
v1.0.2-222-g4a2751c

sonar
v0.0.2-20-gbf5f889

mainsail
v2.2.1

All up to date. 

Log Files

Raabi91, 2022-08-21:
hey i'm sorry to hear that. 

the process is calculated by the file bytes. that some slicers write their data at the beginning and some at the end.
now it may be that it is the filename.
that it doesn't resolve correctly and that's why it doesn't get any data.

if you go to

http://<printer_ip>/printer/objects/query?print_stats&virtual_sdcard&display_status&gcode_move&extruder=target,temperature&heater_bed=target,temperature

in the browser
you will find the tag filename behind it is that the name

take this name and put it at the end of this link 

http://<printer_ip>/server/files/metadata?filename=

and open it once in the browser.

then would be important for me first what you see as output. how the file is in the first window. and how the 2 link looks like if it was automatically resolved in the browser (special characters are automatically converted there)

Stooovie, 2022-08-22:
Hi, thank you, this is the result of the second URL: 

`Satsana_Ender_3_-_5015_Fan_Duct_-_Left_BLTouch_Mount_7h24m_0.24}mm_200C_PLA_ENDER3.gcode `

Stooovie, 2022-08-22:
And this is the result of the first URL, if it's also important: 

`{"result": {"eventtime": 56809.414431245, "status": {"print_stats": {"filename": "Satsana_Ender_3_-_5015_Fan_Duct_-_Left_BLTouch_Mount_7h24m_0.24}mm_200C_PLA_ENDER3.gcode", "total_duration": 8025.529266885998, "print_duration": 7922.347367759001, "filament_used": 7331.974150000173, "state": "printing", "message": ""}, "virtual_sdcard": {"file_path": "/home/pi/gcode_files/Satsana_Ender_3_-_5015_Fan_Duct_-_Left_BLTouch_Mount_7h24m_0.24}mm_200C_PLA_ENDER3.gcode", "progress": 0.27005477344240497, "is_active": true, "file_position": 2557150, "file_size": 9469005}, "display_status": {"progress": 0.27005477344240497, "message": null}, "gcode_move": {"speed_factor": 1.5, "speed": 3000.0, "extrude_factor": 1.05, "absolute_coordinates": true, "absolute_extrude": true, "homing_origin": [0.0, 0.0, -0.025, 0.0], "position": [99.797, 105.432, 18.655, 7597.601500000043], "gcode_position": [99.797, 105.432, 18.68, 1.5116799999997879]}, "extruder": {"target": 200.0, "temperature": 199.98}, "heater_bed": {"target": 60.0, "temperature": 60.01}}}}`

Raabi91, 2022-08-27:
thanks for the info you had the } special character in your gcode name. the browser can resolve it curl but not i have now added something so it should work with the last update

please give me a feedback

Stooovie, 2022-08-30:
I'm happy to report I am getting percentages correctly! The curly brace is added by SuperSlicer for whatever reason... I'll investigate that. But your integ is working fine.

closed 2022-09-11 by Raabi91