Raabi91/moonraker-telegram #112: create_variables failing

BenJuan26, 2023-03-02:
If I request `/state`, at least while a print is not running. I get nothing back from Telegram. Looking at the service logs, it's getting hung up on line 76 of `scripts/actions.sh`, where it's comparing `file_position` to `gcode_end_byte`. Looking at the API response for `objects/query`, there is no field named `gcode_end_byte`. Is this a regression caused by a Klipper/Moonraker update or something?

| Component | Version |
| - | - |
|fluidd|v1.23.2|
|klipper|v0.11.0-122-ge6ef48cd|
|moonraker|v0.8.0-23-g18f5ff4|
|moonraker-telegram|v1.0.2-237-geff5d14|

Raabi91, 2023-03-05:
I will have a look at it. Under query `objects/query` `gcode_end_byte` is not to be found there only the `filename` is pulled. it is retrieved under `/server/files/metadata?filename=$filename`

but currently I can't see anything at mainsail

`gcode_end_byte` is important for a more accurate time calculation because of the thumbmails

BenJuan26, 2023-03-08:
I'm looking at the `create_variables` function, and I don't understand how this ever worked when filename is empty. `$file` will be `""`, which means statements like `echo "$file" | grep -oP '(?<="gcode_start_byte": )[^,]*'` will make no sense at all. Am I missing something here? Shouldn't `create_variables` be skipped if the state is something like `standby`?

closed 2023-05-15 by stale[bot]