Raabi91/moonraker-telegram #114: gif shows picture but state does not

Poikilos, 2023-06-08:
- Version: [8a751ea](https://github.com/Raabi91/moonraker-telegram/commit/8a751ea1e659971e6509de456f19289dd047d247)

## Expected behavior
`/state` should show a picture with the following (default) conf (only relevant part of conf shown):
```
picture="0"
#in this report you can turn off individual images for the individual status. to set no pic set it to "0"
pic_start="1"
pic_error="1"
pic_pause="1"
pic_end="1"
pic_state="1"
pic_standby="1"
pic_complete="1"
pic_paused="1"
pic_bed_cooldown="1"
```

## Actual behavior
`/gif` generates a gif animation, but `/state` does not show a picture.

Related (?):
- The documentation says the conf file is telegram_config.sh, but it is actually named telegram_config.conf (as generated during install).

Poikilos, 2023-06-10:
`~/printer_data/logs/moonraker.log` shows no errors in moonraker-telegram, and the url in the conf is correct (tested by pasting into browser)

Raabi91, 2023-06-10:
Sr,y for late replay you have set picture to 1 to enable images global.

Poikilos, 2023-06-10:
Works. Thanks!

closed 2023-06-10 by Poikilos