Raabi91/moonraker-telegram #118: Placeholders do not work

mtmello2003, 2024-02-26:
With my Elegoo neptune 4 plus, both placeholders:
| $layers          | Total Layers                                    |
| $current_layer   | Current Layer
Do not work:

[image]

msg_state="Printing $print_filename at $print_progress. Current Time $print_current. Remaining Time $print_remaining. Bed $bed_temp/$bed_target. E0 $extruder_temp/$extruder_target. Z $z_current. Layer: $current_layer / $layers."

Raabi91, 2024-03-05:
it works here without problems. try to reinstall bc which we need for the layer calculation.

sudo apt-get install bc

mtmello2003, 2024-03-05:
I did first sudo apt-get update
and after: 
mks@mkspi:~$ sudo apt-get install bc
Reading package lists... Done
Building dependency tree
Reading state information... Done
bc is already the newest version (1.07.1-2+b1).
0 upgraded, 0 newly installed, 0 to remove and 239 not upgraded.

Did nothing...